vmware-archive/cascade #33: Trying to compile on MacOS results in macro error

ajsutrave, 2018-11-12:
I'm seeing this when trying to do a fresh make on MacOS High Sierra. I don't know if it matters, but I installed my dependencies with brew instead of ports. Here is the specific error:

```
ccache g++ --std=c++14 -Werror -Wextra -Wall -Wfatal-errors -pedantic -Wno-overloaded-virtual -Wno-deprecated-register -march=native -fno-exceptions -fno-stack-protector -O3 -DNDEBUG -Iext/googletest/googletest/include -I. -I./ext/cl -c src/runtime/runtime.cc -o src/runtime/runtime.o
In file included from src/runtime/runtime.cc:46:
In file included from ./src/verilog/parse/parser.h:40:
In file included from ./src/verilog/parse/lexer.h:38:
verilog.tab.hh:1001:82: fatal error: too many arguments provided to function-like macro invocation
      basic_symbol (typename Base::kind_type t, YY_RVREF (std::pair<Identifier*, Maybe<RangeExpression>*>) v...
                                                                                 ^
verilog.tab.hh:74:10: note: macro 'YY_RVREF' defined here
# define YY_RVREF(Type)    Type&&
         ^
1 error generated.
make: *** [src/runtime/runtime.o] Error 1
```

DepthDeluxe, 2018-11-13:
This error is coming from a source file generated by bison as the source file gets generated on build.

I get this same error with msys2 gcc 7.3.0 and bison 3.2.  I looked at the OSX homebrew bison package and it uses bison 3.2.1 which is the latest (released 2018-11).

@ajsutrave the project builds successfully if you use either bison 3.0.4 or 3.1.  It must be some issue with the new minor version release.

Would be great if someone could figure out why bison 3.2.* is causing the error message.

eschkufz, 2018-11-13:
I think the issue is that YY_RVREF is a macro. The preprocessor don't know anything about templates, so it's interpreting the comma in the templated type as an argument separator. Essentially, it's parsing the statement as:
```
YY_VREF((std::pair < T1) , (T2>))
```
rather than 
```
YY_VREF((std::pair<T1, T2>));
```

We can fix this with typedefs:
```
typedef std::pair<T1, T2> temp_name
```
and then refer to the typedef everywhere else so that when bison gets around to codegen we end up with
```
YY_RVREF(temp_name)
```

I'll update the FAQ at the end of the README with notes about which versions of bison work until this gets sorted out.

ajsutrave, 2018-11-15:
Downgrading bison worked for me. Let us know if you do end up typedefing the template

eschkufz, 2018-11-15:
Will do --- reopening this ticket as a fix request now that we know what's going on.

eschkufz, 2018-11-30:
(Fixed as of most recent pull request.)

PR #39 (2018-11-30): Support bison 3.1.2
## Overview

Description: This PR fixes an issues with bison 3.1.2, where the build fails due to
```
fatal error: too many arguments provided to function-like macro invocation
```

As documented in issue #33, this is because Bison 3.1.2 uses a new `YY_RVREF` macro, which passes the arguments to the preprocessor. Unfortunately, the preprocessor doesn't know about types and thinks the comma in the std::pair is the second argument, resulting in the too many arguments error.

This PR adds typedefs to replace the usage of std::pair, fixing the issue. 

Related issue(s) (if appli [... cut]
closed 2018-11-30 by eschkufz