vmware-archive/cascade #36: Compiler Error using --march sw

theishshah, 2018-11-29:
## Overview

- ( ) **Bug report.** 

[image]

When following the instructions to use the software fpgs as the backend (from the readme) the entire program crashes and ends.

eschkufz, 2018-11-30:
Can you say a little bit more about your environment? Is the sw_fpga running? Did it emit any error messages? Do you not have read/write permission on /tmp? The two programs will attempt to communicate with each other over a unix domain socket in that directory.

eschkufz, 2018-12-06:
Unable to reproduce. Will reopen if new information becomes available.

closed 2018-12-06 by eschkufz