vmware-archive/cascade #44: Makefile not detecting changes in header files

harshgo, 2018-12-04:
## Overview

When I make a change in a .h file and try to remake Cascade, it does not detect the changes.

**Repro steps**

1. Go the the top level directory 
2. Make a change in `src/base/bits/bits.h`
3. Run `make`

I get `Nothing to be done for 'all'.` I expected the project to be recompiled.

eschkufz, 2018-12-04:
See #43 (which for some reason, the CI is taking endlessly to ok). There should be dependencies on all of the header files in addition to the newly added bison and flex source for the %.o:%.c and %.o %.cc rules.

Probably the cleanest way to do this is to define a new variable similar to ```OBJ``` and add a dependency of that.

eschkufz, 2018-12-06:
Alternately, we can finally migrate to something non-totally-ad-hoc like cmake and all of this should just work.