vmware-archive/cascade #45: Cascade crashes when setting a parameter to itself

kroq-gar78, 2018-12-05:
## Overview

- **Bug report.** When declaring a parameter to be equal to itself, Cascade segfaults. While the semantics for this operation are not well defined, Cascade should either proceed quietly or output an error.

If we try to do this at the REPL, we get some mixed signals:
```
>>> parameter A=A;
ITEM OK
>>> Segmentation fault (core dumped)
```

eschkufz, 2018-12-08:
Great! I'll write up some pointers in a little bit. Cascade has some AST traversal code that will make this pretty straightforward.

eschkufz, 2018-12-08:
Oh! And good news! This is from the spec: 

> The list_of_param_assignments shall be a comma-separated list of assignments, where the right-hand side of the assignment shall be a constant expression, that is, an expression containing only constant numbers and previously defined parameters (see Clause 5).

My reading on this is that placing a variable within its down declaration is illegal. We can just emit an error in typechecking and not worry about having to do any transformations in elaboration.

eschkufz, 2018-12-08:
I just did a fresh checkout of master and verified the bug (though my gdb output looks a little bit different than yours, so make sure you're working with the latest version of the code). 

The file that you'll want to look at is ```src/verilog/program/type_check.cc``` which contains the class ```TypeCheck```. ```TypeCheck``` is like many things in the cascade code base, built around AST traversal. The AST (abstract syntax tree) is the core data structure that we use to store the user's program, and there are four types of traversal classes, of which ```TypeCheck``` is the simplest, a ```Visitor```. ```Visitor```s are responsible for defining methods named ```visit``` for each of the major node types in the AST. You can see these overrides on lines 86-111 of the header file. The method that you're interested in is ```TypeCheck::visit(const ParameterDeclaration* pd) override``` which is defined on line 537 of the corresponding source file ```src/verilog/program/type_check.cc```. As the name suggests, this method is invoked on every parameter declaration in the user's program. I say it's the simplest, because it gives you const access to the node. All you can do is look at it. No modification is allowed.

The definition of the ParameterDeclaration class is in the file ```src/verilog/ast/types/parameter_declaration.h```, which is a sub-type of ```src/verilog/ast/types/declaration.h```. There's a bit of macro-magic going on in these files to avoid us having to write a huge amount of boilerplate code, but hopefully it's easy to see through. The attributes that you care about in the class are ```id``` the variable on the lefthand side and ```val``` the expression on the righthand side. You can access these variables using correspondingly named getters ```ParameterDeclaration::get_val()``` and ```ParameterDeclaration::get_id()```. If you're interested in the details, you can check out ```src/verilog/ast/types/macro.h``` to see how all of this gets generated, but this should be all you need to know.

More or less, what we want to do in ```TypeCheck::visit(const ParameterDeclaration* pd)``` is check whether the variable on the lefthand side of the assignment appears on the righthand side and emit and error. This sounds a bit onerous, but cascade has some code that makes this pretty easy. Something like this should suffice:

```c++
// Grab the set of identifiers which appear in val 
// (ReadSet is defined in src/verilog/analyze/read_set.h)
ReadSet rs(pd->get_val()); 
// Iterate over those variables
for (auto i = rs.begin(), ie = rs.end(); i != ie; ++i) {
  // Resolve this variable. Resolution will give you a pointer to the point in the AST where 
  // a variable is declared, or nullptr if a declaration can't be found. 
  const auto* r = Resolve().get_resolution(*i);
  // If we put this code at the bottom of TypeCheck::visit(const ParameterDeclaration* pd) we 
  // can assume that the checker will already have bailed out if it encountered an unresolvable 
  // identifier.
  assert(r != nullptr);

  // If resolution points us to the lefthand side, this is a recursive definition
  if (r == pd->get_id()) {
    // This method logs an error and a location in the AST to include with the message.
    error("INFORMATION ERROR MESSAGE", pd);
  }
}
```

Once you've fixed this bug, you should probably also go and do the same for LocalparamDeclarations (the same thing looks like it's happening there). The method to look into is ```void TypeCheck::visit(const LocalparamDeclaration* ld);```.

We don't have to worry about GenvarDeclarations since syntactically we aren't allowed to declare and define them in the same line. We also don't have to worry about RegDeclarations since if you try to type ```reg r = r``` the typechecker will yell at you for a different reason. 

There is one last type of declaration that we do need to worry about, and that's NetDeclarations. These are a little bit less intuitive, though. For no particularly good reason other than it being the way the codebase evolved, NetDeclarations get desugared into NetDeclarations followed by ContinuousAssignments during parsing. This means that when you type:

```verilog
wire x = x;
```

(which does produce a segfault) you end up with this by the time you hit the typechecker:

```verilog
wire x;
assign x = x;
```

This isn't a huge deal. It just means that instead of looking in the obvious place ```TypeCheck::visit(const NetDeclaration* nd```, you'll want to look in ```TypeCheck::visit(const ContinuousAssign* ca)```. Here, instead of the lefthand side and righthand sides respectively being accessed with ```pd->get_id()``` and ```pd->get_val()```, you can find them by using the accessors ```ca->get_assign()->get_lhs()``` and ```ca->get_assign()->get_rhs()```. The general idea is the same, though. Iterate over the read set for the righthand side, check whether any of the ids resolve to the lefthand side, emit an error if they do.

That should do it! There was some discussion in one of the recent tickets about how to insert new testcases into the cascade's regression suite, so make sure to add tests for all three cases to make sure everything works. Hopefully that gives you enough to get started. Let me know if anything is unclear and I'll help out however I can. 

Good luck!

eschkufz, 2018-12-12:
Oh, sure. That makes total sense!

The check for the continuous assign case is a little bit different. In this case we want to check whether any of the identifiers on the right hand side resolve to the same variable that the reference on the left hand side resolves to. 

This is a more general case of what we did with parameters and localparams. Though if you think about it, it reduces to what we did there. In the parameter/localparam cases, we knew a priori that resolving the identifier on the left hand side was unnecessary since the resolution would point right back to that same variable. We had the invariant 

```c++
pd->get_id() == Resolve().get_resolution(pd->get_id())
```

In the continuous assign case, we don't have that invariant, since we're working with an assignment which appears somewhere other than where the variable was declared. Adding an additional resolution should do the trick:

```c++
ReadSet rs(ca->get_assign()->get_rhs());
const auto* l = Resolve().get_resolution(ca->get_assign()->get_lhs());
assert(l != nullptr);

for (auto i = rs.begin(), ie = rs.end(); i != ie; ++i) {
  // Resolve the identifier
  const auto* r = Resolve().get_resolution(*i);
  assert(r != nullptr);

  // If it resolves to the left-hand side, this is a recursive definition
  if (r == l) {
    error("Cannot assign a wire to itself", ca);
  }
}
```

Now, there's another (unrelated) bug lurking here which this is going to help expose. Cascade doesn't really have a great way of reasoning about arrays. It treats them as monolithic entities rather than as collections of separate variables. What this means is that your change is going to end up producing an error for the following code:

```verilog
wire x[7:0];
assign x[1] = x[0]; // both left and right hand sides resolve to the same location
```

There's already an open ticket about this ( issue #54 ), so we'll deal with the consequences there. For now, I think this is the right fix to be making.

kroq-gar78, 2018-12-12:
Resolving the LHS seems to work, but I had some problems with the `nullptr` assertions, which I had to demote to if statements instead (i.e. `if (r != nullptr)` ).

```c++
  const auto* l = Resolve().get_resolution(ca->get_assign()->get_lhs());
  if ((l != nullptr) && (dynamic_cast<const NetDeclaration*>(l->get_parent()) == nullptr)) {
    error("Continuous assignments are only permitted for variables with type wire", ca);
  }

  assert(l != nullptr);
```

That last assertion fails for the case `reg x=x;` , so I put the self-assignment logic in an if statement checking that `l != nullptr`. I'm a little confused about 2 things here: 1) since the assertion is being hit, this means a `ContinuousAssign` is being created for `reg`, yet I thought that was only defined for `wire`s; 2) why is `l` a `nullptr` here?

I had another problem with test case `fail_issue_23a`. The type checker _should_ emit an here, but it's not doing so where I expected. I used to have this in `void TypeCheck::visit(const ContinuousAssign* ca)`:

```c++
  const auto* l = Resolve().get_resolution(ca->get_assign()->get_lhs());
  if (l != nullptr) {
    // Iterate over identifiers in the RHS
    ReadSet rs(ca->get_assign()->get_rhs());
    for (auto i = rs.begin(), ie = rs.end(); i != ie; ++i) {
      // Resolve the identifier
      const auto* r = Resolve().get_resolution(*i);

      // If it resolves to the left-hand side, this is a recursive definition
      assert(r != nullptr);
      if (r == l) {
        error("Cannot assign a wire to itself", ca);
      }
    }
  }
```

In that test case, `assert(r != nullptr)` causes an uncaught error, so we fail the test case. Apparently, checking for not-yet-resolved identifiers is done somewhere else, but it seems like we can catch some cases here. I've also demoted this assertion to an if statement for now, since I feel like this is a design decision for you.

Last thing: When you split `wire x=x;` into `wire x;` and `x=x;`, the second statement has no line number associated with it. I'm not sure if this is always a problem if there is an error in the second statement. (If it is, I'll submit another bug report for that.) This is my output for that case:
```
>>> wire x=x;
ITEM OK
>>> *** Typechecker Error:
  > In <unknown location --- please submit bug report> on line 0:
    Cannot assign a wire to itself: assign x = x;
```

I've updated my branch at [kroq-gar78/cascade@issue-45](/kroq-gar78/cascade/tree/issue-45), and I'm planning to submit a PR in a little bit.

eschkufz, 2018-12-13:
This looks great!

You're right about the assertions being demoted to conditionals. Lines 454-458 in your patch are precisely where we recursively descend on the elements of the continuous assign and check for things like unresolvable variables. There's no bailout condition after these recursive invocations so that's why the assertion is too strong. The conditions you added are more or less like saying "I can't check this because something failed earlier on in this method, I'm going to assume that this is an error too".

The thing about the line numbers... meh... the line reporting mechanism we have in the master branch is really clunky. I wouldn't worry about it for the time being. We'll get to it in another ticket.