vmware-archive/cascade #52: Supporting Macros

harshgo, 2018-12-09:
## Supporting macros

While working on our project, I felt that it would have been really useful for us to be able to use Macros in the form of `define`.

eschkufz, 2018-12-13:
This feels tough. I think we'd need a real flex/bison expert to figure this out. I can see how we'd parse macro definitions, but I'm not sure how we'd do the textual replacement in the lexer. I definitely see the utility, though.

eschkufz, 2018-12-13:
Actually, maybe it's not *that* bad. I guess the place to do it would be before the lexer. When we hand the parser a stream, we could scan the entire thing into a string buffer, do the text replacement, wrap it back up in a new stream and then pass it on down to the lexer.

eschkufz, 2019-05-07:
I have some cycles, and this is something I've wanted to do for a while, so I'll go ahead and take it on.

While we're here, there are a few other things that are worth doing. We currently support a niche form of include statements (they come from a part of the spec that deals with IDE-style project management). Let's switch over to supporting directive style includes so we can idiomatically support guarded includes.

eschkufz, 2019-05-09:
All done. The error handling is a bit clunky, but we can deal with that in subsequent threads. Updated the readme to reflect all of the new features this ticket supports.

PR #130 (2019-05-10): Issue 52
## Overview

Description: 

Related issue(s) (if applicable): Fixes #52 

## Checklist (Definition of Done):

- [x] There are no TODOs left in the code
- [x] Change is covered by automated tests
closed 2019-05-10 by eschkufz