vmware-archive/cascade #53: Cascade (sometimes) crashes when multiple loops have the same name

kroq-gar78, 2018-12-09:
## Overview

**Bug report.** Commit `0b7d46fd` .

In certain instances, when loops are assigned the same name, Cascade crashes. In all of the following test cases, the typechecker should catch the error. The first 2 test cases demonstrate unhandled crashes, whereas the last 2 show some sort of inconsistency for when it crashes.

### Test case 1: Two non-empty loops

```verilog
module DUMMY(input wire[31:0] in);

endmodule

genvar i;
for(i = 0; i < 2; i=i+1)
begin : loop_0
    wire [31:0] num;
    DUMMY dummy(num);
end

for(i = 0; i < 2; i=i+1)
begin : loop_0
    wire [31:0] num2;
    DUMMY dummy(num2);
end

always @(posedge clock.val) begin
    $finish(0);
end
```

In debug mode, Cascade crashes with the message:
```
>>> cascade: ./src/base/undo/undo_map.h:121: cascade::BaseUndoMap<K, V, H, E>::const_iterator cascade::BaseUndoMap<K, V, H, E>::insert(K, V) [with K = const cascade::Identifier*; V = cascade::ModuleDeclaration*; H = cascade::HashId; E = cascade::EqId; cascade::BaseUndoMap<K, V, H, E>::const_iterator = std::__detail::_Node_const_iterator<std::pair<const cascade::Identifier* const, cascade::ModuleDeclaration*>, false, true>]: Assertion `map_.find(k) == map_.end()' failed.
Aborted (core dumped)
```

### Test case 2: Empty loop followed by non-empty loop

```verilog
module DUMMY(input wire[31:0] in);

endmodule

genvar i;
for(i = 0; i < 2; i=i+1)
begin : loop_0
end

for(i = 0; i < 2; i=i+1)
begin : loop_0
    wire [31:0] num2;
    DUMMY dummy(num2);
end

always @(posedge clock.val) begin
    $finish(0);
end
```

In debug mode, Cascade crashes with the message:
```
>>> cascade: src/verilog/transform/de_alias.cc:141: virtual void cascade::DeAlias::AliasTable::visit(const cascade::ContinuousAssign*): Assertion `rlhs != nullptr' failed.
Aborted (core dumped)
```

### Test case 3: Non-empty loop followed by empty loop

```verilog
module DUMMY(input wire[31:0] in);

endmodule

genvar i;
for(i = 0; i < 2; i=i+1)
begin : loop_0
end

for(i = 0; i < 2; i=i+1)
begin : loop_0
    wire [31:0] num2;
    DUMMY dummy(num2);
end

always @(posedge clock.val) begin
    $finish(0);
end
```

This case _doesn't_ cause a crash. This is in contrast with the previous test case, where the order of the loops did cause a crash.

### Test case 4: Two loops without modules

```verilog
module DUMMY(input wire[31:0] in);

endmodule

genvar i;
for(i = 0; i < 2; i=i+1)
begin : loop_0
    wire [31:0] num;
end

for(i = 0; i < 2; i=i+1)
begin : loop_0
    wire [31:0] num;
end

always @(posedge clock.val) begin
    $display("This doesn't cause an error. Only module instantiations cause errors.");
    $finish(0);
end
```

This also doesn't cause a crash. It appears that if we don't use modules in these loops, it does not crash.

eschkufz, 2018-12-10:
These are GREAT bug reports!

We have a mechanism for catching this in the typechecker, so this should be easy to fix. The will-it/won't-it crash behavior you see which seems to depend on whether or not you instantiate a module probably has to do with invariants on unique names breaking downstream during code generation.

eschkufz, 2018-12-19:
This ended up being one step removed from an easy fix. We have a mechanism in the typechecker to identify duplicate uses of identifiers (whether they be variables or scope names) which is sensitive to subscripts. The trouble with this issue is that we want to look for duplicate names in a subscript-insensitive fashion. If we try to instantiate a loop with elements 1, 3, 5, or none for that matter,  we need to be able to find a loop instantiated with different indices.

PR #66 (2018-12-19): Fixes issue 53

## Overview

Description: 

Related issue(s) (if applicable): Fixes #53 

## Checklist (Definition of Done):

- [x] There are no TODOs left in the code
- [x] Change is covered by automated tests
closed 2018-12-19 by eschkufz