vmware-archive/cascade #62: Clobbering .mem input files

harshgo, 2018-12-13:
@kroq-gar78 and I found this while working on our project. The last few bits seem to get clobbered when we read in a .mem file *but only sometimes* and *we have absolutely no idea what this condition is*

**Steps to Reproduce**

1. Create a file called `logistic_in.mem` containing:
```
009666 660066 666600 333333 000666 660099 999a00 600000 002ccc cc0003 333300 cccccd 005666 6600a9 999a00 3ccccc 00b999 9a0056 666600 a33333 003ccc cc00a3 333300 6ccccd 003000 000006 6666
```
2. Create a file called `minimal.v` containing:
```verilog
(*__file="logistic_in.mem"*)                                                    
Memory#(5, 32) in_mem(.clock(clock.val), .wen(0), .raddr1(), .rdata1(), .raddr2(), .rdata2(), .waddr(), .wdata());
```

**Result**
When you open `logistic_in.mem` after this operation, you will see the following:
```
9666 660066 666600 333333 666 660099 999a00 600000                                 
2ccc cc0003 333300 cccccd 5666 6600a9 999a00 3ccccc                                
b999 9a0056 666600 a33333 3ccc cc00a3 333300 6ccccd                                
3000 6 0 0 0 0 0 0
```

Looking carefully, you will notice that the last few 6s get clobbered.

We have already checked to make sure that the memory module is large enough to hold all our values.

eschkufz, 2018-12-13:
Cascade will write the contents of a memory back to the file system when your program shuts down. Is your program modifying the state of the memory (overwriting the trailing 6s) while it's running?

harshgo, 2018-12-13:
We originally encountered this behavior while trying to write some code to read in a matrix so that it can be multiplied by some values. However, we have isolated the lines of code which were reading things in (`minimal.v`, see above), and we are still encountering this behavior. Is this expected?

harshgo, 2018-12-15:
No worries!
I pulled today because I thinking of taking a look, and funnily enough I cannot reproduce the test case above anymore as of d7bfdc5a.
However, there is another test case:
```
00a3333300700000002ccccc00066666009ccccc00600000002ccccc00066666009666660066666600299999000666660093333300633333003000000006666600a0000000733333002ccccc00066666
```
After  being passed through `minimal.v` becomes 
```
66666 0 0 0 0 0 0 0                                                             
0 0 0 0 0 0 0 0                                                                 
0 0 0 0 0 0 0 0                                                                 
0 0 0 0 0 0 0 0
```
but this does not happen if there are spaces after every 6 characters in the .mem file.

Does there exist a spec for `.mem` files?

eschkufz, 2018-12-15:
Ah. Whitespace separated hexademical words. In this case it looks like
you're providing one long string, so only the first so many bytes are being
parsed and everything else is being ignored.

eschkufz, 2019-01-30:
This turned out to be pretty simple. When you boot up a software memory it will try to fill every element with the contents of the file you provide. There's an early bailout condition which tells the memory to stop reading when it hits an eof. But if you provide fewer elements than the memory can hold, it hits an eof when it reads the last element and exits before saving that value. Moving the bailout to the other side of the save fixes this.

PR #71 (2019-01-30): Fixes file reading bug for software memories.

## Overview

Description: 

Related issue(s) (if applicable): Fixes #62 

## Checklist (Definition of Done):

- [x] There are no TODOs left in the code
- [x] Change is covered by automated tests
closed 2019-01-30 by eschkufz