vmware-archive/cascade #83: support for yosys backend? (or fill out "adding new backends" docs section?)

shapr, 2019-02-13:
## Overview

I would love to use this with yosys, is that possible?

**Feature request.** 

I've been failing to learn Verilog/FPGA dev and would like something closer to a python REPL.
This looks like the project for me! Since I'm doing this only for fun, I'm using yosys and FPGAs supported by yosys.
Are there plans to add support for yosys? 
Or is there a plan to fill out the documentation section on adding support for new backends, so I could add support for yosys myself?

## Deliverables

- [ ] support for yosys in place of quartus

eschkufz, 2019-02-15:
Yes! This is both possible and on our short list of goals. I'm planning on getting started on both in a few days. When I do, I'll use this ticket to keep track of the progress, so you can keep an eye on what's happening.

eddiehung, 2019-03-01:
I spent a few hours this afternoon hacking together a very proof of concept for the icebreaker board, using Yosys + nextpnr: https://github.com/FPGeh/cascade/tree/icebrk
I have no idea what `program_logic.v` actually does or require, but it basically gets implemented onto the UP5K [with some maneuvering to keep within its 39 I/Os].(https://github.com/FPGeh/cascade/blob/02bc58122853d3de444c2fba76248e666c885900/src/target/core/icebrk/fpga/top.v#L11-L19)

Basically, it's a copy-pasta of the de10 target, but instead of running Quartus it runs Yosys followed by nextpnr, and then icepack to generate the bitstream. I don't have the hardware so [I didn't bother with the final programming step](https://github.com/FPGeh/cascade/blob/02bc58122853d3de444c2fba76248e666c885900/src/target/core/icebrk/icestorm_server.cc#L157-L159).

To get it working, I'll either need to RTFM and see how cascade uses the de10 target, or if there was support from upstream potentially this could be a lot easier.

As an example, I've attached the [output](https://github.com/vmware/cascade/files/2921455/icestorm_server.log) from `icestorm_server` with timestamps so you can see how quickly the whole flow runs. To save you the 2MB click -- for [run_disjunct_16.v](https://github.com/FPGeh/cascade/blob/icebrk/data/test/benchmark/regex/run_disjunct_16.v) it starts at 15:47:01 and has a bitstream by 15:47:09...

eschkufz, 2019-03-04:
Hey, Eddie! Good to see you on the issue tracker!

We were about to expense some yosys-compatible hardware of our own and start hacking, but if you've already gotten started, that's great.

My gut feeling is that the de10 backend code isn't going to work on your device out of the box. The big incompatibility is most likely the mechanism by which the hardware talks back and forth with cascade. In the case of the de10, we take advantage of the fact that there's a dual arm core that can communicate with the fabric over a memory mapped i/o bus. Depending on the i/o peripherals you have connected to your board, you'll probably need to do something different.

Apropos of this ticket title, the 'fm' is currently a combination of my mental model of cascade, our asplos paper which hasn't been published yet, and comments strewn throughout the codebase. So I'm afraid you won't have much luck 'rt'ing it. I'm going to bubble this to the top of my stack and try to write something up so that we can close this ticket and move the discussion to a yosys-backend-specific ticket.

I'll write up some documentation, and put together some skeleton code in the repository so you can see all of the places where you can copy-paste, and where you'll have to spin up something platform-specific.

eddiehung, 2019-03-05:
Thanks for the feedback @eschkufz! Let me rephrase -- what I built isn't even a proof-of-concept, but proof-of-existence/feasibility -- I think the point I wanted to get across most was that we can go from Cascade-generated Verilog to bitstream in a matter of seconds, which I would /assume/ is much more of the timescale you would expect for hardware REPL.

Indeed, what I managed to figure out from the DE10 target is that it invokes QSys and `program_logic.v` is but an IP core on the SoC bus. Obviously, such SoC infrastructure doesn't exist on the icebreaker board, so a rethink is necessary. Currently, I'm not best placed to understand what sort of infrastructure is required (e.g. a small command and control interface, or one that requires lots of data to be shipped back and forth) which also determines whether the icebreaker is the right target to use (it may be a good one to shoot for in some cases, such as teaching).
If you want a Raspberry Pi + ice40 for a standalone system untethered to an x86, then an [icoboard](http://icoboard.org/) may be interesting, though I'm not sure how easy those are to buy nowadays.

eschkufz, 2019-03-15:
Closing this ticket ---

Will reopen a yosys-specific ticket after we have a chance to chat next week.

closed 2019-03-15 by eschkufz

shapr, 2019-08-17:
Was another yosys specific ticket created? or was there "how to add a new backend" documentation created?

eschkufz, 2019-09-03:
Hey Shae ---

There should be some new documentation toward the end of the README in the
master branch of the repository, though it's still a little bit thin.

Eddie and I were talking about standardizing on a yosys-compatible board
and co-developing a backend, though we haven't gotten past the planning
phase yet. Any interest in getting involved?