vmware-archive/cascade #87: Bug: Signed wire interpreted as unsigned in $display

SolitaryThinker, 2019-03-11:
## Overview
Assigning an unsigned wire or bit select to a signed wire overrides the typing for $display(). Causes signed wire to be wrongly interpreted as unsigned for printing purposes.

**Bug report:** 
Commit: a51b778

Example:
```
wire [15:0] unsigned_wire = -1;
wire signed[15:0] signed_wire = unsigned_wire;

initial begin
  $display("wire bits: %d", unsigned_wire);
  $display("signed_wire bits: %d", signed_wire);

  $display("wire bits: %b", unsigned_wire);
  $display("signed_wire bits: %b", signed_wire);
end

>>> wire 65535
>>> signed_wire 65535
>>> wire bits: 1111111111111111
>>> signed_wire bits: 1111111111111111
```

If this is the intended behavior of Verilog please correct me.

eschkufz, 2019-03-11:
This is intended behavior. 

Sign is a property of variables, rather than a function of the 1 or 0 value of their highest-order bit. Verilog has some pretty involved rules for determining whether a variable preserves its sign in one type of statement or another. But in this case:

wire signed x = y; // where y is declared unsigned

The value of x is treated as unsigned.

closed 2019-03-11 by eschkufz

SolitaryThinker, 2019-03-16:
Found the rules in the specs. Thanks!