vmware-archive/cascade #121: Add polling interface to sockserver class

eschkufz, 2019-04-15:
## Overview

sockserver's and sockstream's do a pretty decent job of hiding the ugly c-details of sock communication behind a streaming interface. But there are definitely still places where this abstraction breaks down. Notably, if you have a socket server, you probably want to do polling rather than block in a call to accept. And as soon as you do that, you're back to writing some ugly c-style select loop. 

Let's add a c++-style lambda-based interface to sockservers for running listen loops.

## Deliverables

- [ ] select-style polling interface for sockserver

no2chem, 2019-06-04:
Perhaps we should look into grpc (or grpc with flatbuffers)? Designing our own RPC library sounds like it would be a lot of work.

eschkufz, 2019-07-09:
Closing this for the time being. We have two places in our code where we use sockservers and we use them two different ways, so it's not clear to me what common functionality we'd be refactoring out. If and when we end up with more instances of sockservers, we can revisit this topic.

closed 2019-07-09 by eschkufz