vmware-archive/cascade #122: Cascade hangs on instantiating a large number of array elements

ynaffitgnay, 2019-04-17:
## Overview
When instantiating more than a certain number of elements within a single array, Cascade hangs after the initial declarations in the REPL. This problem occurs whether the instantiations occur inside an initial block or an always block. The instantiated members can be discontiguous, and the error still occurs. When fewer members of an array of the same size are instantiated, the program proceeds as normal.

The memory consumed by Cascade consistently increases by 100 MB every 5 or so seconds while Cascade is hanging. 

A stripped down module that produces this error can be accessed below:
[https://github.com/ynaffitgnay/aos-final/blob/master/bugreports/longarray.v](url)

eschkufz, 2019-04-18:
Confirmed. 

Depending on the size of the array, cascade goes from returning control immediately to hanging indefinitely when the size increases by one, which makes me think this isn't a memory issue.

eschkufz, 2019-04-18:
Oh! I know what it is! 

There *is* a small-bit constant floating around in the simulator. I remember thinking to myself at one point: 'we won't ever have a sequential block with more than 256 items in it'. Well... here we are. This'll be a quick fix.

PR #123 (2019-04-18): Issue 122
## Overview

Description: 

Related issue(s) (if applicable): Fixes #122 

## Checklist (Definition of Done):

- [x] There are no TODOs left in the code
- [x] Change is covered by automated tests
closed 2019-04-18 by eschkufz