vmware-archive/cascade #124: Generate loop fails for decrementing iteration

ynaffitgnay, 2019-04-19:
## Overview

Generate loop fails when decrementing genvar (i.e. iterating over loop from K to 0 vs. incrementing from 0 to K). Cascade crashes during module declaration. This issue does not require urgent attention.

A simplified module to produce the error can be accessed below:
[https://github.com/ynaffitgnay/aos-final/blob/master/bugreports/decrement_generate.v](url)

eschkufz, 2019-04-19:
For readers without read access to tiffany's repository. The code is below:
```verilog
module backwards_generator();
  parameter K = 8;

  genvar i;
  for (i = K - 1; i >= 0; i = i - 1) begin
    if (i == K - 1) initial $display("Started generating");
    else if (i == 0) initial $display("Finished generating");

    wire[31:0] xi;
  end

endmodule // backwards_generator

backwards_generator bg();
```

The bug is pretty straight-forward. The spec says genvar's are to be treated as integers (which are signed). We treat them as unsigned integers, so the loop variable wraps around to 2^32-1 rather than -1 and we hang.

eschkufz, 2019-04-19:
Kind of an odd artifact of this fix: you can address into loop generate constructs with negative indices. I can't find anything in the spec that says this is invalid code, so I guess we're okay. By the time any of this stuff gets to the backend, it'll have long since been renamed away, so we won't have to worry about strange interactions with off-the-shelf compilers.

```verilog
genvar i;
for (i = -1; i >= -2; i=i-1) begin : LOOP
  reg r;
end
initial $display("this works!? %d", LOOP[-1].r);
```

PR #125 (2019-04-19): Fixes 124

## Overview

Description: 

Related issue(s) (if applicable): Fixes #124
## Checklist (Definition of Done):

- [x] There are no TODOs left in the code
- [x] Change is covered by automated tests
closed 2019-04-19 by eschkufz