vmware-archive/cascade #126: De10 Board hangs on the NW code.

SolitaryThinker, 2019-04-20:
## Overview

The entire de10 board will hang as soon as Cascade attempts to migrate to hardware after the quartus compilation. CPU through ssh is unresponsive as well, requiring a manual reset of the board.

The copy of the code that causes this behavior can be found here: [https://github.com/SolitaryThinker/needleman-wunsch/tree/bug/de10](https://github.com/SolitaryThinker/needleman-wunsch/tree/bug/de10
)

The fifo folder contains the code using fifomem for output.
The original_bug contains code with the fifo output replaced with register which causes de10 to hang.

I'm currently looking through my code again for weird corner cases I might have missed. I'll post updates.

Will

SolitaryThinker, 2019-04-20:
Seems like simply removing the FIFO output is enough to produce this behavior.

eschkufz, 2019-06-03:
Closing this ticket as we've deprecated the use of memories and fifo's. The nw benchmarks work with get() and put() now.

closed 2019-06-03 by eschkufz