vmware-archive/cascade #131: Invalid source path causes unit tests to hang indefinitely

no2chem, 2019-05-10:
## Overview

Bug Report - If the unit tests in the test directory pass a file that doesn't exist to ``harness.cc``'s run_code function, the test loops indefinitely instead of producing an error

## Deliverables

- [ ] Produce an error if an invalid path is specified

eschkufz, 2019-05-14:
I think it used to :/. There's something in the googletest api for checking something like that.

eschkufz, 2019-05-14:
Oh... I see what happened. It used to be that we opened a file through the googletest api and then checked whether it existed. What's happening now is we pass cascade an include statement, the statement results in a parse error, and the test never finishes. 

I have a ticket I'm about to open up for improving cascade as a library support. It'll let us check whether an include statement fails and we can use it here to bail out of a test that can't find its source code.

eschkufz, 2019-06-01:
Fixed in issue 132.

closed 2019-06-01 by eschkufz