vmware-archive/cascade #134: Cascade::Parser seems to be missing success member

no2chem, 2019-05-14:
If compiled under debug, ```remote_runtime.cc``` fails too compile:

```
/Users/mwei/devel/cascade/src/target/common/remote_runtime.cc:231:12: error: no member named 'success' in 'cascade::Parser'
  assert(p.success());
         ~ ^
```

The relevant line is here:
https://github.com/vmware/cascade/blob/552e2f1c95f44da2029e309aeab9a0f6902c5669/src/target/common/remote_runtime.cc#L231

eschkufz, 2019-05-14:
I think I got rid of that method when I overhauled the parser. You can
change that line to:
assert(!log.error());

eschkufz, 2019-05-14:
Do we have a debug build on travis? Or at least a build that tries to
compile without -DNDEBUG? That would catch all these errors, no?

no2chem, 2019-05-14:
The cmake branch in #133 should do that now when building and running coverage. (Will be fixed shortly).

eschkufz, 2019-05-14:
Okay to close?

closed 2019-05-15 by eschkufz