vmware-archive/cascade #170: Build fails: member reference type 'std::istream *' (aka 'basic_istream<char> *') is a pointer; did you mean to use '->'

yurivict, 2019-10-21:
```
/usr/ports/cad/cascade-compiler/work/.build/verilog_lexer.cc:1158:8: error: member reference type 'std::istream *' (aka 'basic_istream<char> *') is a pointer; did you mean to use '->'?
                        yyin.rdbuf(std::cin.rdbuf());
                        ~~~~^
                            ->
/usr/ports/cad/cascade-compiler/work/.build/verilog_lexer.cc:1165:9: error: member reference type 'std::ostream *' (aka 'basic_ostream<char> *') is a pointer; did you mean to use '->'?
                        yyout.rdbuf(std::cout.rdbuf());
                        ~~~~~^
                             ->
/usr/ports/cad/cascade-compiler/work/.build/verilog_lexer.cc:2418:50: error: member reference type 'std::istream *' (aka 'basic_istream<char> *') is a pointer; did you mean to use '->'?
                        YY_CURRENT_BUFFER_LVALUE->yy_input_file = yyin.rdbuf();
                                                                  ~~~~^
                                                                      ->
/usr/ports/cad/cascade-compiler/work/.build/verilog_lexer.cc:2537:2: error: cannot initialize a member subobject of type 'std::istream *' (aka 'basic_istream<char> *') with an rvalue of type 'basic_streambuf<std::__1::basic_ios<char, std::__1::char_traits<char> >::char_type, std::__1::basic_ios<char, std::__1::char_traits<char> >::traits_type> *' (aka 'basic_streambuf<char, std::__1::char_traits<char> > *')
        yyin(arg_yyin ? arg_yyin->rdbuf() : std::cin.rdbuf()),
        ^    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/usr/ports/cad/cascade-compiler/work/.build/verilog_lexer.cc:2538:2: error: cannot initialize a member subobject of type 'std::ostream *' (aka 'basic_ostream<char> *') with an rvalue of type 'basic_streambuf<std::__1::basic_ios<char, std::__1::char_traits<char> >::char_type, std::__1::basic_ios<char, std::__1::char_traits<char> >::traits_type> *' (aka 'basic_streambuf<char, std::__1::char_traits<char> > *')
        yyout(arg_yyout ? arg_yyout->rdbuf() : std::cout.rdbuf())
        ^     ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/usr/ports/cad/cascade-compiler/work/.build/verilog_lexer.cc:2540:2: error: use of undeclared identifier 'ctor_common'
        ctor_common();
        ^
/usr/ports/cad/cascade-compiler/work/.build/verilog_lexer.cc:2545:14: error: out-of-line definition of 'yyFlexLexer' does not match any declaration in 'yyFlexLexer'
yyFlexLexer::yyFlexLexer( std::istream& arg_yyin, std::ostream& arg_yyout ):
             ^~~~~~~~~~~
/usr/ports/cad/cascade-compiler/work/.build/verilog_lexer.cc:30:25: note: expanded from macro 'yyFlexLexer'
    #define yyFlexLexer yyFlexLexer
                        ^~~~~~~~~~~
/usr/ports/cad/cascade-compiler/work/.build/verilog_lexer.cc:2536:27: note: type of 1st parameter of member declaration does not match definition ('std::istream *' (aka 'basic_istream<char> *') vs 'std::istream &' (aka 'basic_istream<char> &'))
yyFlexLexer::yyFlexLexer( std::istream* arg_yyin, std::ostream* arg_yyout ):
                          ^
/usr/ports/cad/cascade-compiler/work/.build/verilog_lexer.cc:2546:2: error: cannot initialize a member subobject of type 'std::istream *' (aka 'basic_istream<char> *') with an rvalue of type 'basic_streambuf<std::__1::basic_ios<char, std::__1::char_traits<char> >::char_type, std::__1::basic_ios<char, std::__1::char_traits<char> >::traits_type> *' (aka 'basic_streambuf<char, std::__1::char_traits<char> > *')
        yyin(arg_yyin.rdbuf()),
        ^    ~~~~~~~~~~~~~~~~
/usr/ports/cad/cascade-compiler/work/.build/verilog_lexer.cc:2547:2: error: cannot initialize a member subobject of type 'std::ostream *' (aka 'basic_ostream<char> *') with an rvalue of type 'basic_streambuf<std::__1::basic_ios<char, std::__1::char_traits<char> >::char_type, std::__1::basic_ios<char, std::__1::char_traits<char> >::traits_type> *' (aka 'basic_streambuf<char, std::__1::char_traits<char> > *')
        yyout(arg_yyout.rdbuf())
        ^     ~~~~~~~~~~~~~~~~~
/usr/ports/cad/cascade-compiler/work/.build/verilog_lexer.cc:2549:2: error: use of undeclared identifier 'ctor_common'
        ctor_common();
        ^
/usr/ports/cad/cascade-compiler/work/.build/verilog_lexer.cc:2554:19: error: out-of-line definition of 'ctor_common' does not match any declaration in 'yyFlexLexer'
void yyFlexLexer::ctor_common()
                  ^~~~~~~~~~~
/usr/ports/cad/cascade-compiler/work/.build/verilog_lexer.cc:2592:19: error: out-of-line definition of 'switch_streams' does not match any declaration in 'yyFlexLexer'
void yyFlexLexer::switch_streams( std::istream& new_in, std::ostream& new_out )
                  ^~~~~~~~~~~~~~
/usr/include/FlexLexer.h:129:31: note: type of 1st parameter of member declaration does not match definition ('std::istream *' (aka 'basic_istream<char> *') vs 'std::istream &' (aka 'basic_istream<char> &'))
        virtual void switch_streams( FLEX_STD istream* new_in, FLEX_STD ostream* new_out = 0 );
                                     ^
/usr/include/FlexLexer.h:53:22: note: expanded from macro 'FLEX_STD'
#    define FLEX_STD std::
                     ^
/usr/ports/cad/cascade-compiler/work/.build/verilog_lexer.cc:2596:41: error: no viable conversion from 'std::istream' (aka 'basic_istream<char>') to 'std::istream *' (aka 'basic_istream<char> *')
        yy_switch_to_buffer( yy_create_buffer( new_in, YY_BUF_SIZE  ) );
                                               ^~~~~~
/usr/include/FlexLexer.h:121:62: note: passing argument to parameter 's' here
        struct yy_buffer_state* yy_create_buffer( FLEX_STD istream* s, int size );
                                                                    ^
/usr/ports/cad/cascade-compiler/work/.build/verilog_lexer.cc:2599:7: error: member reference type 'std::ostream *' (aka 'basic_ostream<char> *') is a pointer; did you mean to use '->'?
        yyout.rdbuf(new_out.rdbuf());
        ~~~~~^
             ->
/usr/ports/cad/cascade-compiler/work/.build/verilog_lexer.cc:2607:12: error: assigning to 'std::istream *' (aka 'basic_istream<char> *') from incompatible type 'std::istream **' (aka 'basic_istream<char> **'); remove &
                new_in = &yyin;
                         ^~~~~
/usr/ports/cad/cascade-compiler/work/.build/verilog_lexer.cc:2611:13: error: assigning to 'std::ostream *' (aka 'basic_ostream<char> *') from incompatible type 'std::ostream **' (aka 'basic_ostream<char> **'); remove &
                new_out = &yyout;
                          ^~~~~~
/usr/ports/cad/cascade-compiler/work/.build/verilog_lexer.cc:2614:17: error: no viable conversion from 'std::istream' (aka 'basic_istream<char>') to 'std::istream *' (aka 'basic_istream<char> *')
        switch_streams(*new_in, *new_out);
                       ^~~~~~~
/usr/ports/cad/cascade-compiler/work/.build/verilog_lexer.cc:2604:49: note: passing argument to parameter 'new_in' here
void yyFlexLexer::switch_streams( std::istream* new_in, std::ostream* new_out )
                                                ^
/usr/ports/cad/cascade-compiler/work/.build/verilog_lexer.cc:2623:11: error: member reference type 'std::istream *' (aka 'basic_istream<char> *') is a pointer; did you mean to use '->'?
        if ( yyin.eof() || yyin.fail() )
             ~~~~^
                 ->
/usr/ports/cad/cascade-compiler/work/.build/verilog_lexer.cc:2623:25: error: member reference type 'std::istream *' (aka 'basic_istream<char> *') is a pointer; did you mean to use '->'?
        if ( yyin.eof() || yyin.fail() )
                           ~~~~^
                               ->
fatal error: too many errors emitted, stopping now [-ferror-limit=]
```

bison-3.4.1,1
FreeBSD 12

eschkufz, 2019-10-21:
This error appears when compiling against an older version of flex. Cascade requires flex 2.6.0 or newer.

eschkufz, 2020-01-03:
Won't fix, but let's at least make the build script a bit more helpful.

yurivict, 2020-01-03:
I have ```flex-2.6.4``` and ```bison-3.4.2``` satisfying the requirement in CMakeLists.txt but I am still getting the same failure.

The versions that you require are wrong, or there's some other problem.

eschkufz, 2020-01-03:
Can you try pulling the most recent commit to master, rm -rf'ing your build
directory, running the setup script again, and then copying its output?

I wonder if you have multiple instances of flex installed on your machine
and the script isn't finding the correct one to use.

eschkufz, 2020-01-03:
(Reopening while we figure this out --- I really want to add freebsd as a stable build target)

yurivict, 2020-01-03:
I do use the very latest commit.

yurivict, 2020-01-03:
No multiple versions:
```
-- Found FLEX: /usr/local/bin/flex (found suitable version "2.6.4", minimum required is "2.6.0")
-- Found BISON: /usr/local/bin/bison (found suitable version "3.4.2", minimum required is "3.2.0")
```

yurivict, 2020-01-03:
> I really want to add freebsd as a stable build target

Thanks!

yurivict, 2020-01-03:
Maybe it's all the way around and you are using too old versions? I use the latest flex-2.6.4, and nearly the latest bison-3.4.2.

Why don't you upgrade to the latest flex/bison?

eschkufz, 2020-01-03:
I'm on the same versions. Something else must be going on.

eschkufz, 2020-01-03:
Okay. This seems promising ---

We have the same version of flex, but it's possible that your copy of flex didn't install an up-to-date copy of FlexLexer.h.

Try looking through gcc's include paths for FlexLexer.h. You can find these paths by typing:

```bash
cpp -xc++ -v < /dev/null
```

When you find the file (there may be more than one --- gcc will use whichever one it finds first), look through the file for the definition of yyFlexLexer::yyin. In my copy of FlexLexer.h, it's defined like this:

```c++
std::istream yyin;
```

I bet that in your copy of FlexLexer.h it'll be defined like this:

```c++
std::istream* yyin;
```

yurivict, 2020-01-03:
I figured out what's going on. On BSD base system comes with the older flex and its header ```/usr/include/FlexLexer.h``` is included before the newer one ```/usr/local/include/flex/FlexLexer.h``` installed by the flex port.

I don't think you can do anything about this, this is just an idiosyncrasy in FreeBSD. I'll just patch it in the port.

closed 2020-01-03 by yurivict

eschkufz, 2020-01-03:
I'll update the README with notes.

PR #204 (2020-01-03): Assert minimum flex version in cmake.

## Overview

Description: 

Related issue(s) (if applicable): Fixes #170 

## Checklist (Definition of Done):

- [x] There are no TODOs left in the code
- [x] Change is covered by automated tests