vmware-archive/cascade #214: cascade attempts to open globally installed files without prefix

yurivict, 2020-01-04:
```
$ cascade
>>> Parse Error:
>>>   > Unable to locate file share/cascade/march/sw.v
>>>   > In final line of user input:
>>>     syntax error, unexpected <unparseable>
>>> Caught Signal
```

```$ ln -s /usr/local/share``` helps as a quick workaround, but you need to use installation prefix when you open files.

eschkufz, 2020-01-04:
This is probably related to your other ticket about the regression tests failing. 

By default, cascade will try to resolve includes against your installation directory (```/usr/local/``` for example) and the current working directory. The way it gets this path at runtime is kind of convoluted: it looks up cascade's location in your file system and then goes up a level. So if it finds cascade in ```/usr/local/bin/```, it will assume that cascade is installed in ```/usr/local```.

This all happens in ```System::src_root()```, which is found in ```src/cascade/common/system.h```. If I had to guess, I would bet that this method isn't working correctly on your machine and it's returning an empty string.

Here's an easy way to check. Try checking out my most recent commit to issue-214, rebuilding your code, and running ```cascade --enable_info```. Right after it says "Started logical simulation" you'll now see a line that prints out what cascade thinks the installation directory is. If there's nothing there, or the path looks wrong, we'll know that's our problem, and that ```System::src_root()``` is where to look.

yurivict, 2020-01-04:
This PR fixes the problem of global path on BSDs: https://github.com/vmware/cascade/pull/220

But regression tests still fail.

eschkufz, 2020-01-04:
Closing this ticket. We'll pick back up in the regression test failing ticket.

closed 2020-01-04 by eschkufz