vmware-archive/cascade #229: Documentation on how tobring up DE10 Nano board?

yurivict, 2020-01-06:
The README says:
```
[...] you'll need an SD card image for your DE10 with a valid installation of Cascade. Cascade can generate this image for you automatically [...]
```
The obvious question is "how". Is there a command that generates the image?

I have a working DE Nano, it has linux running, it is connected through the Ethernet.
Is this image not sufficient, and Cascade should generate another image, as the above README paragraph seems to suggest?

Could you please add relevant information to the README file, because in its current form it is lacking information needed to use Cascade on DE10 Nano.

(I think it is possible to compile and install a driver on the existing DE10 image that would have 100% access to FPGA. This is how they have their fft demos.)

eschkufz, 2020-01-06:
Sorry about that --- Improving our README is on my short list of things to do for the new year.

Short answer: 

A while ago when we acquired some de10s, they came with a copy of angstrom linux installed on their sd cards. At the time, cascade had some dependencies which we couldn't satisfy with opkg, so one of my colleagues put together a stripped down image with ubuntu and a copy of cascade on it. That's the image that I'm referring to in the README. I'm out of office today, but when I go in tomorrow, I'll grab a copy and get it uploaded.

Longer answer:

I need to check in with my colleague about what the steps he used to build the image were. Once I have those, I can post them as well. Alternately, it *could be* that cascade builds on angstrom now (I haven't tried in a while --- I know the setup script won't work, but it should just be a matter of getting the dependencies satisfied and running cmake like on any other *nix target). Might be worth a quick try in the meantime.

eschkufz, 2020-01-09:
Update ---

I've got the disk image, but it's 8GB, which is larger than what I can upload to github. I don't have any experience distributing files this large, do you have a suggestion for how to host it?

yurivict, 2020-01-09:
@eschkufz  Thanks, but I have a better solution in mind now.

I will submit a PR soon once I will make it work.

eschkufz, 2020-01-09:
Can't wait!