vmware-archive/cascade #230: mmap is called with length=1 but more than 1 bytes is later used, this must be wrong

yurivict, 2020-01-07:
Here:
https://github.com/vmware/cascade/blob/master/src/cascade/target/core/avmm/de10/de10_config.cc#L83
the second argument (len) is 1, but the returned memory array is used as if it is larger, for example here https://github.com/vmware/cascade/blob/master/src/cascade/target/core/avmm/de10/de10_config.cc#L118 the byte at offset ```CTRL_OFFSET=4``` is used.

This must be wrong. What is the correct len?

eschkufz, 2020-01-09:
That certainly *seems* wrong to me. That bit of code was adapted from here:

https://github.com/nhasbun/de10nano_fpga_linux_config/blob/master/main.c

Your guess is as good as mine as to what the correct len should be.

eschkufz, 2020-01-10:
It seems like the correct value for length should be ```GPIO_INTSTATUS+1```